loboris/MicroPython_ESP32_psRAM_LoBo #292: sdcard fails

supersjimmie, 2019-03-27:
Plain and simple 1GB SDCard, fresh (full) formatted to fat32 on my laptop.
It's placed on the esp32 pcb, with notes on the bottom of the pcb:
```
TF    ESP32
CS   IO13
MOSI   IO15
SCK   IO14
MISO   IO2
```
I also placed this in the menuconfig, where I choose SPI.
```
>>> import uos
>>> uos.sdconfig(uos.SDMODE_SPI, clk=14, mosi=15, miso=2, cs=13)
>>> uos.mountsd(True)
E (37522) sdmmc_cmd: sdmmc_card_init: sdmmc_send_cmd_crc_on_off returned 0x106
E (37523) vfs_native: Failed to initialize SDcard (262).
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
```

supersjimmie, 2019-03-27:
With SPI I would expect SPI or 4LINE to work, but it seems that only 1LINE works?
```
>>> uos.sdconfig(uos.SDMODE_4LINE)
>>> uos.mountsd(True)
E (6279) sdmmc_cmd: sdmmc_send_cmd_switch_func: sdmmc_send_cmd returned 0xffffffff
E (6280) vfs_native: Failed to mount filesystem on SDcard.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO

>>> uos.sdconfig(uos.SDMODE_1LINE)
>>> uos.mountsd(True)
---------------------
 Mode:  SD (1bit)
     Name: 00000
     Type: SDSC
    Speed: high speed (50 MHz)
     Size: 971 MB
      CSD: ver=0, sector_size=512, capacity=1989632 read_bl_len=9
      SCR: sd_spec=2, bus_width=5
```

lyusupov, 2019-08-01:
I have **no** issues with an SD card both in 1 and 4 line modes:

```
>>> import uos
>>> uos.sdconfig(uos.SDMODE_1LINE)
>>> uos.mountsd()
---------------------
 Mode:  SD (1bit)
     Name: SMI
     Type: SDHC/SDXC
    Speed: high speed (50 MHz)
     Size: 3841 MB
      CSD: ver=1, sector_size=512, capacity=7866368 read_bl_len=9
      SCR: sd_spec=2, bus_width=5

>>> uos.listdir('/sd')
['vhusbdmips', 'logcfg.out', 'log0001.csv', 'log0002.csv', 'log0003.csv', 'log0]
>>> uos.umountsd()
>>> uos.sdconfig(uos.SDMODE_4LINE)
>>> uos.mountsd()
---------------------
 Mode:  SD (4bit)
     Name: SMI
     Type: SDHC/SDXC
    Speed: high speed (50 MHz)
     Size: 3841 MB
      CSD: ver=1, sector_size=512, capacity=7866368 read_bl_len=9
      SCR: sd_spec=2, bus_width=5

>>>
```

My ESP32 pins wiring (TTGO T8 V1.8) does match to the reference one:
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/filesystems#sd-card-connection-to-esp32

closed 2019-08-09 by supersjimmie