loboris/MicroPython_ESP32_psRAM_LoBo #296: error architecture not supported on mac os

hyhkjiy, 2019-05-30:
➜  MicroPython_BUILD git:(master) ✗ ./BUILD.sh -v

---------------------
MicroPython for ESP32
---------------------

================================
Building MicroPython firmware...
================================
/Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/xtensa-esp32-elf/bin:/Users/jiy/.pyenv/plugins/pyenv-virtualenv/shims:/Users/jiy/.pyenv/shims:/Users/jiy/.pyenv/plugins/pyenv-virtualenv/shims:/Users/jiy/.pyenv/shims:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/usr/local/go/bin:/usr/local/opt/mysql-connector-c/include:/usr/local/opt/mysql-connector-c/include
Building partitions from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
CC build/app_trace/app_trace.o
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:32:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/_types.h:27,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/string.h:61,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/app_trace.c:158:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/cdefs.h:784:2: error: #error Unsupported architecture
 #error Unsupported architecture
  ^
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:33:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/_types.h:27,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/string.h:61,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/app_trace.c:158:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/machine/_types.h:34:2: error: #error architecture not supported
 #error architecture not supported
  ^
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/_types.h:27:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/string.h:61,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/app_trace.c:158:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:55:9: error: unknown type name '__int64_t'
 typedef __int64_t       __darwin_blkcnt_t;      /* total blocks */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:56:9: error: unknown type name '__int32_t'
 typedef __int32_t       __darwin_blksize_t;     /* preferred block size */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:57:9: error: unknown type name '__int32_t'
 typedef __int32_t       __darwin_dev_t;         /* dev_t */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:60:9: error: unknown type name '__uint32_t'
 typedef __uint32_t      __darwin_gid_t;         /* [???] process and group IDs */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:61:9: error: unknown type name '__uint32_t'
 typedef __uint32_t      __darwin_id_t;          /* [XSI] pid_t, uid_t, or gid_t*/
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:62:9: error: unknown type name '__uint64_t'
 typedef __uint64_t      __darwin_ino64_t;       /* [???] Used for 64 bit inodes */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:68:9: error: unknown type name '__darwin_natural_t'
 typedef __darwin_natural_t __darwin_mach_port_name_t; /* Used by mach */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:70:9: error: unknown type name '__uint16_t'
 typedef __uint16_t      __darwin_mode_t;        /* [???] Some file attributes */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:71:9: error: unknown type name '__int64_t'
 typedef __int64_t       __darwin_off_t;         /* [???] Used for file sizes */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:72:9: error: unknown type name '__int32_t'
 typedef __int32_t       __darwin_pid_t;         /* [???] process and group IDs */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:73:9: error: unknown type name '__uint32_t'
 typedef __uint32_t      __darwin_sigset_t;      /* [???] signal set */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:74:9: error: unknown type name '__int32_t'
 typedef __int32_t       __darwin_suseconds_t;   /* [???] microseconds */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:75:9: error: unknown type name '__uint32_t'
 typedef __uint32_t      __darwin_uid_t;         /* [???] user IDs */
         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types.h:76:9: error: unknown type name '__uint32_t'
 typedef __uint32_t      __darwin_useconds_t;    /* [???] microseconds */
[... 321 lines cut ...]
 typedef __darwin_ct_rune_t ct_rune_t;
         ^
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdlib.h:78:0,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/include/freertos/FreeRTOSConfig.h:119,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/include/freertos/FreeRTOS.h:99,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/app_trace.c:166:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types/_rune_t.h:31:9: error: unknown type name '__darwin_rune_t'
 typedef __darwin_rune_t rune_t;
         ^
In file included from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/include/esp_app_trace.h:17:0,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/app_trace.c:167:
/Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/xtensa-esp32-elf/lib/gcc/xtensa-esp32-elf/5.2.0/include/stdarg.h:98:24: error: conflicting types for 'va_list'
 typedef __gnuc_va_list va_list;
                        ^
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/_stdio.h:75:0,
                 from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/stdio.h:64,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/include/esp_err.h:17,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/xtensa-debug-module/include/trax.h:3,
                 from /Users/jiy/work/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/app_trace/app_trace.c:163:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types/_va_list.h:32:26: note: previous declaration of 'va_list' was here
 typedef __darwin_va_list va_list;
                          ^
make[1]: *** [app_trace.o] Error 1
make: *** [component-app_trace-build] Error 2
'make all' FAILED!

hyhkjiy, 2019-05-30:
The reason is that my computer's environment variable is incorrect.

closed 2019-05-30 by hyhkjiy

moyaying, 2020-06-20:
What environment variable?   I face the same problem, can you share me?