loboris/MicroPython_ESP32_psRAM_LoBo #302: Is this project active?

beyonlo, 2019-08-01:
The last commit was 11 months ago. I know that this doesn't mean it project is active or not. But Loboris do not reply the issues too.

I would like that this amazing project will be grown!

At this moment I don't know if a start a new product with Loboris micropython fork or from original micropython.

Anyone has any news about Loboris and evolution of this esp32 micropython port?

Thank you

carterw, 2019-08-01:
It would be great to see loboris comment, but I fear that this project is now abandonware. He is actively committing to his Kendryte K210 project in the repository adjacent to this one, which appears to be focused on low-cost computer vision. 

I suppose one or more enterprising and knowledgeable developers could fork this repository and at least implement bug fixes. Maybe it has already been done?

Another option would be for the 'official' Micropython group to import some of the very nice features that were implemented here. The build system with its many tuneable options is a big plus for example.

nevercast, 2019-08-12:
@carterw Are you saying that the feature you like most is menuconfig?

carterw, 2019-08-12:
There are a lot of great features, but the build system lets you conveniently set a huge number of parameters and burn both the firmware and the filesystem plus contents with just a couple of commands. Not seeing anything close in the main micropython repository.

nevercast, 2019-08-12:
@carterw Perhaps you might be interested in joining our Slack channel for MicroPython? Plenty discussion happens there as far as the direction MicroPython goes. We are discussing board configs currently, re: https://github.com/micropython/micropython/pull/4991

We are certainly interested in implementing good features that have found their way in to Boris's port.

carterw, 2019-08-12:
Would be glad to participate, can I get an invitation?

nevercast, 2019-08-12:
Sure, shall I use your bill@j&b.com email address, or shall I email you asking for the email you wish to have invited, if you want to keep it private.

carterw, 2019-08-12:
That address should work, thanks.

nevercast, 2019-08-12:
Done.

mattytrentini, 2019-08-14:
For reference (and sorry for going off-topic!) anyone can get an [automatic invite](https://slack-micropython.herokuapp.com/) to the MicroPython Slack workspace.

carterw, 2019-08-15:
For convenience [I've put up a repository that just has a README](https://github.com/carterw/loboris_backports) in which we can capture things people think are important.

jdtsmith, 2019-12-28:
Has there been any development on this front?  Trying to decide whether to base a new project on this more feature-rich MP build, or wait for main MP to incorporate...

carterw, 2019-12-28:
Having spoken with some of the mainstream MicroPython contributors on Slack, my sense is that not much of the loboris port will be incorporated. The _thread module was pulled in for ESP32 but that's about it. I've advocated for the build menu and script to be ported over, but there seems to be little interest in it. I've been told that most people just use the precompiled firmware binaries available on the internet and that should generally be adequate.

Part of the problem is that mainstream MicroPython is multi-platform, whereas this port is ESP32-specific. There is disagreement about how to incorporate platform-specific features, some are opposed to it. Also the sole arbiter of what does get included in MicroPython is Damien George, and he apparently has no interest in what was done here.

A motivated person with a lot of technical expertise could probably put together a set of git PR's  that could be submitted to the mainstream repository for review. Maybe some of it would be accepted, but maybe not.

jdtsmith, 2019-12-28:
Useful info, thanks.  Not invented here syndrome sadly rampant.  Some things seem rather separable.  For example, I've quickly grown quite fond of mDNS, telnet and ftp connectivity. The latter pair beats the pants off other options like Webrepl (much faster transfer, for example, and can use ncftpput -z to recursively upload changed files, all at once).  Are there any thoughts to port these standalone add-ons as "modules" to include with standard Micropython builds?  Though I don't know enough to say, I don't suspect they are terribly ESP32-specific.

carterw, 2019-12-28:
I agree those modules are very useful, and it would be great if they were included in the mainstream builds. I don't know where they originally came from. I suggest chiming in on the Slack channel, I find that one lonesome guy there doesn't get much traction.

One problem is that most Micropython platforms do not have built-in WiFi, the ESP32 is one of the few. A lot of the the developers work with STM32 which doesn't even have any inherent connectivity. So that gets us into the platform-specific box, and therefore an argument.

mattytrentini, 2019-12-29:
> Some things seem rather separable. 

In my experience, they're not nearly as separable as I'd hoped. Perhaps the features you want are, I'm not sure. Certainly I had a good look at display, neopixel and RMT but couldn't easily use them.

> For example, I've quickly grown quite fond of mDNS, telnet and ftp connectivity. The latter pair beats the pants off other options like Webrepl (much faster transfer, for example, and can use [...]

Take a look, we'd love to have a PR! I _may_ be able to find time to help though they're not high on my wish-list...
I suspect the issue you'll have with those features is that they're quite large and require connectivity which will rule them out for many ports. Still, it would be useful even if only for the ESP32 port.

> Not invented here syndrome sadly rampant. 

I don't believe that's the case; it's just that a lot of the code isn't _easily_ ported upstream and there are only few developers helping out for ESP32-only features. 

@carterw makes a good suggestion - you may find the [slack channel](https://slack-micropython.herokuapp.com/) a good place to discuss some of this.

jdtsmith, 2019-12-29:
Good thoughts.  In the meantime I've found some work-around solutions with mainline uPy. mDNS is now implemented in the new 1.12 release on ESP32's (though setting the hostname is still undocumented — it's `wlan.config(dhcp_hostname=HOSTNAME)`).  And `utelnetserver` (once suitably patched to function), and `uftpd` (patched to report file modification time), seem to be pretty fast and relatively stable telnet/ftp servers.  Both of these are pure python.  Perhaps the right focus is to improve the system for C-modules to easily be incorporated into uPy.

jedie, 2020-01-15:
Think it's a good idea to port missing features into mainline micropython, see: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/304