loboris/MicroPython_ESP32_psRAM_LoBo #306: Including files in MicroPython.bin

behackful, 2019-08-30:
Hi, I'm trying to solve this thing for 2 days now.
I want to create a firmware.bin that will include some files(main.py, some HTML and css etc) that will be **in the flash memory** but I just can't find a way to do it.

The only way I found, that is not really logical, is to add another function in the vfs_native.c file,  same as the function that checks if "boot.py" exists and creates it for main.py file (or any other file).

I know that I can create an image with the files but the problem is that I need one file because I intend to work exclusively with OTA updates.

Is there something I'm missing? I tried to freeze the required files but it only works for python scripts.

carterw, 2019-09-01:
I haven't experimented with OTA yet but very interested. With ordinary building and flashing you can create two images; 'firmware' and filesystem images that you install onto the device separately. Could you do the same kind of thing with OTA? Looks like the firmware image gets pulled down from a server with FTP and then installed locally. Could the same kind of thing happen with a filesystem image?

nissimzur, 2020-06-02:
Behackful, 
Did you have success?

behackful, 2020-06-02:
> Behackful,
> Did you have success?

Sorry, no.

nissimzur, 2020-06-05:
Enable FTP or telnet (in menu config) and you can add any file you like in run time.

gitk1, 2020-09-05:
Hey guys! did you have any progress on this? I am also having same doubt. I want to implement OTA. I checked from docs that '.bin' can be written on another partition (other than from current). However, this .bin file is the firmware, its not main.py or boot.py (which I want to update). Please correct, if I am wrong here. 
I looked for other methods like using github project etc, but it doesn't look like secure method. Can we put main/boot.py directly in the ota_partition using ota module included in lobo micropython? will that work? Can we include main.py/boot.py in the 'micropython.bin'. [I know that separate file image can be created and flashed as @behackful mentioned ]. @nissimzur could you please elaborate which you mentioned?
Thank you.