loboris/MicroPython_ESP32_psRAM_LoBo #307: ucryptolib

rvignero, 2019-09-03:
Hi, can't find ucryptolib module, any plan to get it on lobo's firmware ? he's implemented in micropython original firmware v1.11.
Thanks

awesomo4000, 2020-02-24:
I know it's been a while but I just merged  ucryptolib into my fork of this repo.
These two commits -
48449d09270a1fb464cdf7e2a96b3019ab62ed52
4eda3e846979a6b85a5531034313d4c5e9ddbf64

Cheers.

chrisraff, 2020-03-03:
@awesomo4000 I added (and double checked) those changes to my files and the build fails:
`make[1]: *** No rule to make target 'py/../xtmod/moducryptolib.c', needed by 'mypath/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/build/genhdr/qstr.i.last'.  Stop.`
Do you know what the issue might be?

awesomo4000, 2020-03-08:
I see  `xtmod` which might need to be `extmod`.