loboris/MicroPython_ESP32_psRAM_LoBo #310: TTGO T-Display ST7789V

k4m454k, 2019-11-21:
[image]

not correctly work
MISO = not connected, but display lib required **miso** parameter, im set to pin 0

```
>>> tft.init(tft.ST7789, rst_pin=23, backl_pin=4, miso=0, mosi=19, clk=18, cs=5, dc=16, width=135, height=240, backl_on=1) 
>>> tft.clear()
>>> tft.text(100,100, 'HUI', tft.YELLOW)
```

[image]

k4m454k, 2019-11-21:
## Pinout
| Name       | V18    |
| ---------- | ------ |
| TFT Driver | ST7789 |
| TFT_MISO   | N/A    |
| TFT_MOSI   | 19     |
| TFT_SCLK   | 18     |
| TFT_CS     | 5      |
| TFT_DC     | 16     |
| TFT_RST    | N/A    |
| TFT_BL     | 4      |

edwardmccaughan, 2019-12-17:
I have also been having the same problems, but got it mostly figured out.

first, the inverted colors problem is solved by someone on the lobo forum. For some reason the color inversion is being set wrong, but there is a low level command to fix it:
`tft.tft_writecmd(0x21)`
https://loboris.eu/forum/showthread.php?tid=188

for the screen only drawing to part of the screen, there is also some weirdness in the offset, so I used `setwin` to 'fix' it. I'm not sure if this is the right way to do it, but it works...

```
tft.init(tft.ST7789,
         rst_pin=23,
         backl_pin=4,
         miso=0,
         mosi=19,
         clk=18,
         cs=5,
         dc=16,
         width=screen_width_with_offset,
         height= screen_height_with_offset,
         backl_on=1,
) 
tft.setwin(width_offset,height_offset, screen_width_with_offset + 100, screen_height_with_offset+100)

ST7735_INVON = const(0x21)
tft.tft_writecmd(ST7735_INVON)
```

The MISO pin does not seem to be important in the end

Hope this helps!

hsph, 2019-12-17:
> I have also been having the same problems, but got it mostly figured out.
>
> first, the inverted colors problem is solved by someone on the lobo forum. For some reason the color inversion is being set [...]

Your approach did the trick for me. Finding out the right window size / offset was bit of trial and erroring. The values below worked for my [TTGO T-Display](https://www.banggood.com/LILYGO-TTGO-T-Display-ESP32-CP2104-WiFi-bluetooth-Module-1_14-Inch-LCD-Development-Board-For-Arduino-p-1522925.html?rmmds=myorder&cur_warehouse=UK).

```
import display

# initialize the display
tft=display.TFT()
tft.init(tft.ST7789, rst_pin=23, backl_pin=4, miso=0, mosi=19, clk=18, cs=5, dc=16, width=235, height=340, backl_on=1)

# invert colors
tft.tft_writecmd(0x21)

# set orientation (optional)
tft.orient(tft.LANDSCAPE)

# set window size
tft.setwin(40, 52, 278, 186)
```

Alancreed, 2020-02-16:
Thanks @hsph! The values you provided helped a lot. Just want to add a minor fix(for setwin parameters):

import display

# initialize the display
tft=display.TFT()
tft.init(tft.ST7789, rst_pin=23, backl_pin=4, miso=0, mosi=19, clk=18, cs=5, dc=16, width=235, height=340, backl_on=1)

# invert colors
tft.tft_writecmd(0x21)

# set orientation
tft.orient(tft.LANDSCAPE)

# set window size
 tft.setwin(40, 52, 279, 186)

These values give the full : 240 * 135 that our display is capable of. This can be confirmed by running:
 tft.winsize()