loboris/MicroPython_ESP32_psRAM_LoBo #313: Adafruit ST7789 - ESP32 DeviKit v4

LDS79, 2020-04-04:
Adafruit ST7789 Wide Angle 240x240
ESP32 DeviKit v4
I'm try to learn how to use the module Display but If I digit this code:

import display
tft = display.TFT()
tft.init(tft.ST7789, miso=19, mosi=23, clk=18, cs=5, dc=27)
tft.clear(tft.GREEN)

the result is a monitor all purple, in addition also the x y coordinates are totally out of range!
if I write:
tft.init(tft.ST7789, miso=19, mosi=23, clk=18, cs=5, dc=27, height=240)
I see only half monitor and the point 0,0 are still out side of monitor..

Where I'm wronging?
Thanks for the help

valentingregoire, 2020-12-24:
I have the exact same. For me, the exact range of the screen is (when landscape/landscape_flip):
* x goes from 40 to 279 instead of 0 to 320
* y goes from 53 to 187 instead of 0 to 240
This results in a total different aspect ratio. I dont't like this.

LDS79, 2021-01-04:
> I have the exact same. For me, the exact range of the screen is (when landscape/landscape_flip):
> 
> * x goes from 40 to 279 instead of 0 to 320
> * y goes from 53 to 187 instead of 0 to 240
>   This results in a total different aspect ratio. I dont't like this.
> 
> Also the colors are off. Black and white are flipped, the rest is just weird. Even if I try with play around with the `bgr` parameter.

LDS79, 2021-01-04:
> I have the exact same. For me, the exact range of the screen is (when landscape/landscape_flip):
>
> * x goes from 40 to 279 instead of 0 to 320
> * y goes from 53 to 187 instead of 0 to 240
>   This [...]

I personally did not find any satisfactory solution and I switched to an OLED...
if you find a solution please keep me updated!

valentingregoire, 2021-01-04:
> I personally did not find any satisfactory solution and I switched to an OLED...
> if you find a solution please keep me updated!

I have figured out how to fix the colours with this comment: 
https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/310#issuecomment-566305438
In this issue, there is more interesting info. I use a [LilyGO ESP32 TTGO](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1126&FId=t3:50033:3), so I couldn't switch the display.

Here is my setup that works just fine:
```python
TFT = display.TFT()
# 320 * 240, but in reality x goes from 40 to 279 and y goes from 53 to 187 -> 239 * 134 --> 1.78 ratio instead of 1.33
# --> 240 * 135
TFT.init(
    TFT.ST7789,
    rot=TFT.LANDSCAPE_FLIP,
    miso=17,
    backl_pin=4,
    backl_on=1,
    mosi=19,
    clk=18,
    cs=5,
    dc=16,
    splash=False,
)
TFT.tft_writecmd(0x21)
TFT.setwin(39, 52, 279, 187)  # 240 * 135
TFT.savewin()
```