loboris/MicroPython_ESP32_psRAM_LoBo #331: Compile Problem at mpy-cross

xpix, 2021-03-24:
$ ./BUILD.sh -v all

---------------------
MicroPython for ESP32
---------------------

==================
Building mpy-cross
FAILED
==================
'make all' FAILED!

**Solution:
pacman -S python**

merdogmus, 2022-03-20:
Thanks for solution

Damon3499, 2022-06-20:
I running on my Mac into the same trouble. How did you fix this?

piyushy6, 2023-01-01:
> I running on my Mac into the same trouble. How did you fix this?

Hi @Damon3499 
Were you able to solve this?

McuTronics, 2024-09-20:
I still can't get it to work

I have followed every instruction and it always fails

I use Ubuntu on Windows 10, the first step of menuconfig does it satisfactorily

but when running ./BUILD all, it fails

On Windows 10, it always fails to unzip .tar files