loboris/MicroPython_ESP32_psRAM_LoBo #336: 'make all' FAILED! ??

getpythonpy, 2022-05-17:
I was building the firmware on Linux for esp32 and when I run BUILD.sh  i keep getting  'make all' FAILED!

piyushy6, 2023-01-01:
> I was building the firmware on Linux for esp32 and when I run BUILD.sh I keep getting 'make all' FAILED!

Hi @getpythonpy 
Were you able to solve this issue?

carterw, 2023-01-01:
I think if you build with the -v option it will print some debugging info that should help you.

McuTronics, 2024-09-20:
> I think if you build with the -v option it will print some debugging info that should help you.

even setting the -v option, it does not show details, it just fails

at least in my case, it is in the last step with the mpy-cross

I think that the current versions of tools cause a problem, without more support to solve or updates it will be difficult to solve it, I think.

McuTronics, 2024-09-20:
I finally did it.

Use Ubuntu version 18.04, it works on Windows (wsl)

Follow the steps as described in the wiki and it works