loboris/MicroPython_ESP32_psRAM_LoBo #340: TTGO - can't use display and another HSPI device at the same time

tttonyyy, 2022-12-21:
Really love the project and have it up and running on a TTGO board with display.

The issue I've got is that I can't use the display and an externally connected SPI thermocouple at the same time.

```
import display
from machine import SPI
hspi = SPI(spihost=SPI.HSPI, sck=25, mosi=33, miso=27, cs=26)
tft = display.TFT()
tft.init(tft.ST7789,bgr=False,rot=tft.LANDSCAPE, miso=17,backl_pin=4,backl_on=1, mosi=19, clk=18, cs=5, dc=16, splash=False)
```

Whichever is initialised first seems to have exclusive use of HSPI.

In the example above, any attempt to access the tft results in clocking on pin 25.

If I init the TFT first then SPI, the display works fine, but a hspi read does not produce any clock signal on pin 25.  (The chip select does assert, though).

I can deinit one and init the other and use them one at a time, but really the chip select should do that for me.

carterw, 2022-12-21:
The project has been dormant for years. You should probably be working with the mainstream Micropython releases.

tttonyyy, 2022-12-21:
Thinking about it (it was late, I was tired) it might be working correctly anyway; sharing SPI requires the devices to use the same clock and miso/mosi pins. I don't think the HSPI pins the display uses are externally exposed.

tttonyyy, 2022-12-21:
The the TFT uses HSPI, but if I attempt to use VSPI while the TFT is initialised, I get the following error:

```
>>> vspi = SPI(spihost=SPI.VSPI, sck=25, mosi=33, miso=27, cs=26)
E (36855689) spi_master: spi_bus_initialize(95): dma channel already in use
E (36855689) [SPI_UTILS]: spi bus initialization failed with rc=0x103
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: invalid state
```

This device has no PSRAM, so VSPI should be available.  I do not get the same errror if the TFT is not initilised.

The TFT driver and hw spi appear to use the same DMA channel even if different SPI interfaces are used.

I modified `components/micropython/esp32/machine_hw_spi.c` to change the channel from 1 to 0 (no DMA - can't use dma channel 2 because that's used for VFS).  My SPI peripheral is low-bandwidth so this is not a problem:
 
```self->spi.dma_channel = 0;```

Now I can use the TFT on HSPI and my thermocouple on VSPI.

closed 2022-12-21 by tttonyyy